jimm98y/SharpOnvif #1: ONVIF Device Test Tool

EK-Reveal, 2024-08-05:
Hi,

I've successfully integrated this library (good work btw) in a small ONVIF device emulator I'm creating, and it's being detected by both ONVIF Device Manager and the free Happytimesoft ONVIF client.

However I'm having problems with the official ONVIF Device Test Tool, the application is not being detected. Has anyone tried this before?

jimm98y, 2024-08-05:
Thanks! Unfortunately, I do not have access to those test tools, but if you figure it out I am willing to accept pull requests.

EK-Reveal, 2024-08-16:
The ONVIF Device Test Tool doesn't send the action in the request (not in the content type, nor in the soap headers), I ended up having to infer it from the soap body. I'm still making some tests, will create a pull request later.

jimm98y, 2024-09-29:
Could it be related to https://github.com/onvif/testspecs/issues/63? So perhaps the new version of the device test tool will fix this issue?

closed 2024-10-07 by jimm98y