jimm98y/SharpOnvif #2: How use Server project

masgh021, 2024-09-24:
Hello @jimm98y 
I want make my pc to onvif device and connect to other device via c# 
can you help ?
what are steps ?

jimm98y, 2024-09-25:
You have to run SharpOnvifService in Visual Studio as Administrator. Then your PC should become discoverable by Onvif Device Manager.

masgh021, 2024-09-26:
Thanks @jimm98y . I tested and my pc is discoverable .

closed 2024-09-26 by jimm98y

masgh021, 2024-09-26:
Hi @jimm98y 
please check this image . 
[image]

 I want to add a sample video for test 
live-video and live-streaming But I can't .  how is it ? 
thanks .

jimm98y, 2024-09-26:
For video stream you need some RTSP url. To get it, you have to use some IP camera, or SharpRealTimeStreaming RTSP server https://github.com/jimm98y/SharpRealTimeStreaming/tree/main/src/SharpRTSPServer. 
Then you have to update https://github.com/jimm98y/SharpOnvif/blob/main/src/OnvifService/Onvif/MediaImpl.cs and return your RTSP url from the GetStreamUri.

masgh021, 2024-09-26:
Thanks @jimm98y 
I download rtsp server source and run it , it gives me this link rtsp://admin:password@127.0.0.1:8554 
I test link in vlc and worked , now I want add this link for live-video or live-streaming option in onvif sharponvif project, I don't know how update function and return this link. could you help ?    
Thanks.

masgh021, 2024-12-13:
Hello @jimm98y 
I couldn't find how send stream url to Onvif device manager. 
do you have any sample ?

jafar88A, 2024-12-13:
Hi @masgh021 
I have same Problem.

jimm98y, 2024-12-13:
You have to put this URL here: https://github.com/jimm98y/SharpOnvif/blob/d2b26dd22b491d5a1a8335bb3e6df3b7c393c862/src/OnvifService/Onvif/MediaImpl.cs#L52C17-L52C61

jimm98y, 2024-12-14:
The server sample now supports RTSP as well as PTZ from Onvif Device Manager.

jimm98y, 2025-01-02:
There were a few more changes needed, not just the URL. See https://github.com/jimm98y/SharpOnvif/pull/15/files#diff-239792789ac25553aa399a86b2a342bcdb73d636a1403eea6f2355af71a0ae95. Please get the latest sources from this repo as they contain updated sample app.

masgh021, 2025-01-03:
Thanks @jimm98y  this problem solved,
but I have new problem and when I connect another pc via LAN to my laptop and I start application but I couldn't  find my laptop in onvif device manager . what is the problem . (I can ping my laptop IP in CMD )

masgh021, 2025-01-16:
Hello @jimm98y 
Do you have a solution?

jimm98y, 2025-01-19:
Sounds like a networking issue. Onvif is using UDP broadcast for discovery and those UDP broadcast datagrams must be able to reach Onvif Server. You should check the IP in which Onvif Server is listening (in case you have multiple network interfaces, make sure it's using the same one as your "client". Also you might want to try disabling Windows Firewall and see if that fixes your problem.

masgh021, 2025-01-23:
Thanks @jimm98y .

eantunes1980, 2025-05-08:
Hi @jimm98y ,

thank you for the great project. 
Unfortunately I'm couldn't getting the Server running. Getting Authentication errors, Internal Server Errors,... 
Can you provide a sample of a Onvif-Server with one RTSP-Stream?

jimm98y, 2025-05-08:
You need to be a more specific - what kind of authentication/internal server errors are you getting? Can you share the stack trace?

The sample with one RTSP stream is already in this repo: https://github.com/jimm98y/SharpOnvif/tree/main/src/OnvifService

The steps to run it locally:
1) clone this repo
2) run Visual Studio as Administrator
3) compile and run https://github.com/jimm98y/SharpOnvif/tree/main/src/OnvifService

If you want to also have RTSP, then:
1) clone https://github.com/jimm98y/SharpRealTimeStreaming
2) run Visual Studio as Administrator
3) compile and run https://github.com/jimm98y/SharpRealTimeStreaming/tree/main/src/RTSPServerApp

Now using Onvif Device Manager, you should see this server and you should also be able to see the RTSP stream.

eantunes1980, 2025-05-08:
Thank you. I've tested the first example it's working. I will try to work with this sample. I havn't shared details because was searching a sample like that. Thank you, if I have other troubles I will share more details ;)

PR #15 (2024-12-14): F RTSP and PTZ in the Onvif Device Manager
closes #2