jimm98y/SharpOnvif #3: problem in getStreemUri

hnor, 2024-09-25:
hi

in axis camera 
client.GetStreamUriAsync return error 400(bad request)

can you help me to resolve this??

tanks

jimm98y, 2024-09-25:
I don't have any Axis camera at my disposal, could you please capture the requests in Wireshark and send me the pcap?

hnor, 2024-09-28:
hi
tanks for your responce
this is your lib xml from wireshark
<s:Envelope xmlns:s=\"http://www.w3.org/2003/05/soap-envelope\">
		<s:Header>
			<Security xmlns=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd\">
				<UsernameToken xmlns:xsi=\"http://www.w3.org/2001/XMLSchema-instance\" xmlns:xsd=\"http://www.w3.org/2001/XMLSchema\">
					<Username>root</Username>
					<Password Type=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest\">cYPzgIieq3DxtCG2yp16Y3W7dio=</Password>
					<Nonce EncodingType=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary\">0ew1ZhKzy73ah0As3wMfgSgMN2ryeOGNOkOZnaHmNFo=</Nonce>
					<Created xmlns=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd\">2024-09-28T17:25:26.954Z</Created>
				</UsernameToken>
			</Security>
		</s:Header>
		<s:Body xmlns:xsi=\"http://www.w3.org/2001/XMLSchema-instance\" xmlns:xsd=\"http://www.w3.org/2001/XMLSchema\">
			<GetStreamUri xmlns=\"http://www.onvif.org/ver10/media/wsdl\">
				<StreamSetup>
					<Stream xmlns=\"http://www.onvif.org/ver10/schema\">RTP-Unicast</Stream>
				</StreamSetup>
				<ProfileToken>profile_1_h264</ProfileToken>
			</GetStreamUri>
		</s:Body>
	</s:Envelope>
	

this is onvif device manager xml from wire shark:
	<s:Envelope xmlns:s=\"http://www.w3.org/2003/05/soap-envelope\">
		<s:Header>
			<Security s:mustUnderstand=\"1\" xmlns=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd\">
				<UsernameToken>
					<Username>root</Username>
					<Password Type=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest\">YYLRRQSXc8BX2KUvxkCfQR3ml2o=</Password>
					<Nonce EncodingType=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary\">pKH86hJs/0SGUEkevSvQJA8AAAAAAA==</Nonce>
					<Created xmlns=\"http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd\">2024-09-28T06:29:10.375Z</Created>
				</UsernameToken>
			</Security>
		</s:Header>
		<s:Body xmlns:xsi=\"http://www.w3.org/2001/XMLSchema-instance\" xmlns:xsd=\"http://www.w3.org/2001/XMLSchema\">
			<GetStreamUri xmlns=\"http://www.onvif.org/ver10/media/wsdl\">
				<StreamSetup>
					<Stream xmlns=\"http://www.onvif.org/ver10/schema\">RTP-Unicast</Stream>
					<Transport xmlns=\"http://www.onvif.org/ver10/schema\">
						<Protocol>RTSP</Protocol>
					</Transport>
				</StreamSetup>
				<ProfileToken>profile_1_h264</ProfileToken>
			</GetStreamUri>
		</s:Body>
	</s:Envelope>	

thank you very much

closed 2024-09-29 by jimm98y

jimm98y, 2024-09-29:
Thank you very much for the Wireshark! Yes, it looks like the transport element was missing. The bug was in SimpleOnvifClient which is an optional component intended to make it easier for the developers to get started, but you can always use the underlying APIs and/or create your own SimpleOnvifClient with a different set of APIs that fit your needs. The issue should be fixed in 0.0.3.