jimm98y/SharpOnvif #6: How call GetStreamUri  Or GetVideoSources function

masgh021, 2024-09-29:
Hi
How call GetVideoSources Or GetStreamUri function to return a url    ?

masgh021, 2024-09-30:
Hi @jimm98y 
I checked codes today but didn't understand  how return url to user and 
I didn't find out Procedures for calling functions .

can you help me ?
Thanks

jimm98y, 2024-11-30:
How to call `GetStreamUri` is shown here: [https://github.com/jimm98y/SharpOnvif/blob/main/src/OnvifClient/Program.cs](https://github.com/jimm98y/SharpOnvif/blob/main/src/OnvifClient/Program.cs)

It's quite simple:
```cs
var client = new SimpleOnvifClient("http://192.168.1.111/onvif/device_service", "admin", "password");
var services = await client.GetServicesAsync(true);
if (services.Service.FirstOrDefault(x => x.Namespace == OnvifServices.MEDIA) != null)
{
    var profiles = await client.GetProfilesAsync();
    var streamUri = await client.GetStreamUriAsync(profiles.Profiles.First());
    
    // do something with streamUri here
}
```

closed 2024-11-30 by jimm98y