jimm98y/SharpOnvif #7: How do I use this library to control PTZ?

geometrikal, 2024-10-09:
Trying to work out how I send PTZ commands to a Dahua camera

wwwlaomao, 2024-10-10:
You may refer to the sample code in src/OnvifClient/Program.cs and check if PTZ is available and then call the PTZ related extension methods on client. The PTZ methods can prototypes can be found in src/SharpOnvifClient.PTZ/Connected Services/OnvifPTZ/Reference.cs.

jimm98y, 2024-11-30:
I suppose something like this might work:

```cs
#region PTZ

public async Task AbsoluteMoveAsync(string profileToken, float pan, float tilt, float zoom, float panSpeed = 1, float tiltSpeed = 1, float zoomSpeed = 1)
{
    string ptzURL = await GetServiceUriAsync(OnvifServices.PTZ).ConfigureAwait(false);
    using (var ptzClient = new PTZClient(
        OnvifBindingFactory.CreateBinding(),
        new System.ServiceModel.EndpointAddress(ptzURL)))
    {
        SetAuthentication(ptzClient.Endpoint, _auth);
        await ptzClient.AbsoluteMoveAsync(
            profileToken,
            new PTZVector()
            {
                PanTilt = new PTZ.Vector2D() { x = pan, y = tilt },
                Zoom = new PTZ.Vector1D() { x = zoom }
            },
            new PTZ.PTZSpeed()
            {
                PanTilt = new PTZ.Vector2D() { x = panSpeed, y = tiltSpeed },
                Zoom = new PTZ.Vector1D() { x = zoomSpeed }
            }).ConfigureAwait(false);
    }
}

public async Task RelativeMoveAsync(string profileToken, float pan, float tilt, float zoom, float panSpeed = 1, float tiltSpeed = 1, float zoomSpeed = 1)
{
    string ptzURL = await GetServiceUriAsync(OnvifServices.PTZ).ConfigureAwait(false);
    using (var ptzClient = new PTZClient(
        OnvifBindingFactory.CreateBinding(),
        new System.ServiceModel.EndpointAddress(ptzURL)))
    {
        SetAuthentication(ptzClient.Endpoint, _auth);
        await ptzClient.RelativeMoveAsync(
            profileToken,
            new PTZVector()
            {
                PanTilt = new PTZ.Vector2D() { x = pan, y = tilt },
                Zoom = new PTZ.Vector1D() { x = zoom }
            },
            new PTZ.PTZSpeed()
            {
                PanTilt = new PTZ.Vector2D() { x = panSpeed, y = tiltSpeed },
                Zoom = new PTZ.Vector1D() { x = zoomSpeed }
            }).ConfigureAwait(false);
    }
}

#endregion PTZ
```

Please let me know if it works for you.

closed 2024-12-07 by jimm98y