jimm98y/SharpOnvif #10: Can not find any device

pubpy2015, 2024-11-23:
I am using your lib on nuget to search ONVIF camera in local network, but it can not find any device. My code:
var onvifDeviceUris = await DiscoveryClient.DiscoverAsync();
foreach(var s in onvifDeviceUris)
{
    listBox1.Items.Add(s);
} 
Onvif Device manager work well
[image]

sumonnwe, 2024-11-25:
I also got same issue from last Friday. Previously it was working.

jimm98y, 2024-11-25:
Are you using it from a NET8 app, or NET Framework app? Could you please provide Wireshark logs of the discovery? In my network it is still working from the OnvifClient test app.

sumonnwe, 2024-11-26:
[image]

Here is my debug screen shot. I am running with OnvfiClient test app . Also in my client program got same error. But I can discover the device using with that library  https://github.com/vicmaeg/onvif-discovery

jimm98y, 2024-11-26:
There haven't been any changes in the discovery since May and the code is pretty self-contained, using just the UDP client. 

I'd suggest trying:
1) Use the IP address of your network interface as the first parameter, e.g. 172.16.0.123. You can get the IP address by calling ipconfig in the cmd line. Call DiscoveryClient.DiscoverAsync("172.16.0.123") (replace 172.16.0.123 with the IP address of your network interface).
2) Use the nuget https://www.nuget.org/packages/SharpOnvifClient version 0.0.3 and see if you can still reproduce the issue. This should rule out any changes in the actual code.

Unfortunately, I am not able to reproduce the issue here, so the only way to actually see what is going on would be if you could provide me with the Wireshark logs of the discovery from Onvif Device Manager and then from the OnvifClient app.

sumonnwe, 2024-11-27:
Hi Jimm98y , thanks for your reply. I could solve yesterday likewise your ways by setting client IP address when discover the device. But previously I don't need to set IP to discover. Anyway, thanks for your reply and the problem is already solved. Now I stuck with PTZClient Digest Header. The error is here. 

System.ServiceModel.Security.MessageSecurityException: The HTTP request is unauthorized with client authentication scheme 'Anonymous'. The authentication header received from the server was 'Digest qop="auth", realm="IP Camera(F1598)", nonce="4e446731597a466b4d7a45364e7a4930593249784e513d3d", stale="FALSE"'.
   at System.Runtime.AsyncResult.End[TAsyncResult](IAsyncResult result)
   at System.ServiceModel.Channels.ServiceChannel.SendAsyncResult.End(SendAsyncResult result)
   at System.ServiceModel.Channels.ServiceChannel.EndCall(String action, Object[] outs, IAsyncResult result)
   at System.ServiceModel.Channels.ServiceChannelProxy.TaskCreator.<>c__DisplayClass2_0.<CreateTask>b__0(IAsyncResult asyncResult)
--- End of stack trace from previous location ---

jimm98y, 2024-11-28:
Looking into the original issue, it looks like the problem can happen on PCs with multiple network adapters. In such case you can't use the default 0.0.0.0 address, but you must get the IP address of the NIC connected to the network you want to search. If you want to search all networks, you have to iterate over all the NICs and issue multiple discovery requests. 0.0.0.0 works fine on PCs with a single NIC, which is why I can't reproduce the issue. Hope it helps. For the other question please create a new issue.

sumonnwe, 2024-11-29:
Thanks @jimm98y  for your explanation. Sure I am working on ETHERNET meanwhile connecting with WiFi. So I pass Ethernet IP when discover IP camera connecting with cable. So it was solved. I will create issue for another case.

PR #12 (2024-11-29): F Added support for multiple NICs
closes #10
closed 2024-11-29 by jimm98y