jimm98y/SharpOnvif #11: PTZ Control got digest authentication error with hikvision camera

sumonnwe, 2024-11-29:
I am using Sharponvifclient and my camera itself as a server so don't run Sharponvifserver. From where, I got an issue when calling the PTZ control. The camera is also enabled onvif protocol and ptz is suppoted. Check with ONVIF DEVICE MANAGER got authentication issue even though I am using the same account of web login as there's no setting for new account specifically for PTZ. When running PTZ relativemoveasync got that error. 

Now I stuck with PTZClient Digest Header. The error is here.

System.ServiceModel.Security.MessageSecurityException: The HTTP request is unauthorized with client authentication scheme 'Anonymous'. The authentication header received from the server was 'Digest qop="auth", realm="IP Camera(F1598)", nonce="4e446731597a466b4d7a45364e7a4930593249784e513d3d", stale="FALSE"'.
at System.Runtime.AsyncResult.End[TAsyncResult](IAsyncResult result)
at System.ServiceModel.Channels.ServiceChannel.SendAsyncResult.End(SendAsyncResult result)
at System.ServiceModel.Channels.ServiceChannel.EndCall(String action, Object[] outs, IAsyncResult result)
at System.ServiceModel.Channels.ServiceChannelProxy.TaskCreator.<>c__DisplayClass2_0.b__0(IAsyncResult asyncResult)
--- End of stack trace from previous location ---

jimm98y, 2024-11-29:
Did you add the `DigestBehavior` to `ptzEndpoint.EndpointBehaviors`? Please see the `SetAuthentication` method in the ` SimpleOnvifClient`.

sumonnwe, 2024-12-02:
[image]

Yes, I added but got protocol error exception and 400 bad request.

jimm98y, 2024-12-02:
Well, if it does not work with Onvif Device Manager, then it will not work with SharpOnvif. It looks to me like your camera does not support the `ContinuousMove` action. You should first query the capabilities of the camera and only call actions that are actually supported.
Please see https://github.com/jimm98y/SharpOnvif/issues/7 for a snippet using the `AbsoluteMove` and `RelativeMove` actions and see if they work with your camera.
If you make it work with Onvif Device Manager and the same call will fail in SharpOnvif, then please capture the calls from Onvif Device Manager as well as SharpOnvif using Wireshark and we can take a look and see what's going on.

sumonnwe, 2024-12-03:
[image]
[image]

Query the capabilities, got the Digest authentication error. 
Check with ODM, got 404 Bad request either. So how can I proceed it? 
Can I change any setting and protocol at my hikvision camera DS-2DE3A404IW-DE/W. Actually I already enabled ONVIF, PTZ control? I am working on MainStream profile. 

[image]

jimm98y, 2024-12-03:
Based off the error, it looks like the camera is forcing Digest in the HTTP Authorization header, while SharpOnvif (and Onvif Device Manager to my knowledge) only support WS-UsernameToken. 
Please see https://ipcamtalk.com/threads/onvif-problem-with-hikvision-dvr-the-http-request-is-unauthorized-with-client-authentication-scheme.42433/post-747891 for more info and perhaps a solution on how to configure the camera.
On SharpOnvif side, supporting Digest Authentication header according to RFC 2617 is an enhancement that would have to be implemented.

closed 2024-12-07 by jimm98y