jimm98y/SharpOnvif #16: QUESTION on Basic Auth support

mikkel75, 2025-01-09:
Hi,

Is it possible to force the Onvif HTTP request to use Basic Auth instead of the XML envelope embedded login/password method?

Thanks.

jimm98y, 2025-01-14:
Are you using the library acting as a server (acting as Onvif device) or as a client (connecting to Onvif device)? The Onvif specification as far as I know does not specify Basic: https://www.onvif.org/specs/core/ONVIF-Core-Specification.pdf#5.9.
The latest version of the library supports the legacy Ws-UsernameToken (XML-embedded as you say) and Digest in the HTTP header (see the PR: https://github.com/jimm98y/SharpOnvif/pull/14). Basic is not supported - do you have an actual device that is using Basic authentication for Onvif?

mikkel75, 2025-01-14:
Using the lib as client, connecting to different AI cameras (Dahua, Bosch...).
I have no problem using Ws XML to Dahua, but so far only only basic auth working to Bosch (it's specifically enabled in Bosch cam).
Do I need to use SetAuthentication before using the SimpleOnvifClient ?
Side question - do the lib support GetMetadataConfigurations and GetAnalyticsConfigurations functions? Do you have an example on how?

jimm98y, 2025-01-15:
Interesting, so Bosch supports Basic authentication for Onvif? Could you please share a screenshot of the configuration and how it's enabled on your camera? Which model of the Bosch camera are you using?

Which version of the library are you using? The latest one should correctly support Digest in HTTP header (RFC 2617) and only use Ws-Usernametoken in case the camera does not support RFC 2617. 

As for Basic Authentication, the only mention of "basic" in the specification that I have found refers to using JWT token for authentication and using the Basic Authentication in the OAuth2 flow. JWT is currently not supported, but it could be added in the future. If you send me more info about your camera, I can look into the way they are supporting Basic.

"do the lib support GetMetadataConfigurations and GetAnalyticsConfigurations functions? Do you have an example on how?"
I don't have an example, but yes, they are supported on both Client and the Server: https://github.com/jimm98y/SharpOnvif/blob/fc322fcb6fd8830c92b014d6fe3c28642fda8db0/src/SharpOnvifClient.Media/Connected%20Services/OnvifMedia/Reference.cs
You should be able to call it the same way as https://github.com/jimm98y/SharpOnvif/blob/main/src/SharpOnvifClient/SimpleOnvifClient.cs - see GetStreamUriAsync implementation.

sweco-nlkristen, 2025-02-12:
Also running into authentication issues here, wondering if no authentication is possible?

ikriz, 2025-02-12:
Any possibilities to configure the Server without Authentication, currently running into issues with the Digest Method:

```
SharpOnvifServer.DigestAuthenticationHandler: Information: Digest was not authenticated. Failure message: No authentication found
SharpOnvifServer.DigestAuthenticationHandler: Information: AuthenticationScheme: Digest was challenged.
SharpOnvifServer.DigestAuthenticationHandler: Information: Digest was not authenticated. Failure message: HTTP Digest authentication failed.
```

jimm98y, 2025-02-12:
How are you calling it?

The errors are expected behavior: client makes a request, server responds with a challenge (www auth) and the next request from the client should contain the correct authentication. 

Other option is to use the Ws-Usernametoken which lets you include the authentication into the Soap message. 

The client from this project should work as well as Onvif Device Manager. Which client are you using?

ikriz, 2025-02-12:
Calling it from the Onvif test tool, it seems the digest compare fails when using admin / password as set in the example.

jimm98y, 2025-02-12:
Hmm that one I don't have so I can't test it. If you would be willing to capture the Wireshark log of the communication with the test app (using default passwords) and share it, I can look into it. There used to be this issue with some older version of the test tool: https://github.com/onvif/testspecs/issues/63

ikriz, 2025-02-12:
I guess i have an old version then

closed 2025-03-02 by jimm98y