jimm98y/SharpOnvif #17: Exception thrown while trying to get profiles for onvif client

carlos-chourio, 2025-03-21:
I am using the code from the Program.cs on the OnvifClient project 

https://github.com/jimm98y/SharpOnvif/blob/main/src/OnvifClient/Program.cs

But when trying to get profiles I get the following exception:

System.InvalidCastException
  HResult=0x80004002
  Mensaje = Unable to cast object of type 'SharpOnvifClient.DeviceMgmt.DeviceClient' to type 'SharpOnvifClient.Media.MediaClient'.
  Origen = SharpOnvifClient
  Seguimiento de la pila:
   en SharpOnvifClient.SimpleOnvifClient.GetOrCreateClient[TClient,TChannel](String uri, Func`2 creator)
   en SharpOnvifClient.SimpleOnvifClient.<GetProfilesAsync>d__17.MoveNext()

carlos-chourio, 2025-03-21:
Checking the code I saw that it is because in the simple client when we request the device information the client gets stored internally in a _clients variable as a DeviceClient, since DeviceClient and MediaClient are of two different types they can not be casted to another. For a fix I checked the type of the variable before the cast in case they were not the same type then create and return the type that is requested.

In line 69 I made this change:

        public TClient GetOrCreateClient<TClient, TChannel>(string uri, Func<string, TClient> creator) where TClient : ClientBase<TChannel> where TChannel : class
        {
            lock (_syncRoot)
            {
                if (_clients.ContainsKey(uri))
                {
                    var typedClient = _clients[uri] as TClient;
                    if (typedClient != null) {
                        return typedClient;
                    }
                    else {
                        return CreateClient<TClient, TChannel>(uri, creator);
                    }
                }
                else
                {
                    var client = CreateClient<TClient, TChannel>(uri, creator);
                    _clients.Add(uri, client);
                    return client;
                }
            }
        }

        private TClient CreateClient<TClient, TChannel>(string uri, Func<string, TClient> creator) where TClient : ClientBase<TChannel> where TChannel : class
        {
            var client = creator(uri);
            client.SetOnvifAuthentication(_authentication, _credentials, _legacyAuth);
            return client;
        }

Maybe it's not the best approach since you are not saving the new created client in the dictionary, but since the uri is the same I can not save it without replacing the old one. Maybe the dictionary could contain a key of uri and also type (like a tuple or so) so it could contain multiple client types per type

closed 2025-03-23 by jimm98y

jimm98y, 2025-03-23:
Thank you very much for reporting this, it should be fixed.

PR #18 (2025-03-23): B Fixed clients with same uri
Closes #17