jimm98y/SharpOnvif #19: I think my devices sucks

brunobittarello, 2025-03-24:
Hey, first, congratulations for this work.

I spent the last 2 days trying to access my both Yoosee cameras using coding. I tried many solutions, which I could list here, in C#, Python and even Javascript, but my 2 cameras could not be accessed. At first I thought I was doing something wrong or those repositories were too old, but then I found this one, updated yesterday.

I can see the stream using rtsp://admin:<<password>>@192.168.0.xx/onvif1 using OBS and VLC. I can do all the stuff using Yoosee app for mobile and windows. I can get the stream in save the video using Python, the only thing I cant do is what I want: move the camera.

Reading #10 I noted the Ovif Device Manager and I download it. Both cameras were detected, but nothing worked well. The strange part is the PTZ control works by moving a fraction of an angle per input, when the input is not invalidated. For one camera, only horizontal moves are accept.

I also tried to detect what Yoosee windows app sent using WireShark, but I could not understand the data. I see some packages were sent directly to the cameras IP.

I am writing this just to check if you have a solution or idea for me to handle this. I guess the only solution is to throw them into garbage and by new ones. 😄

Thanks!

jimm98y, 2025-03-24:
I'd need a Wireshark log of the communication from the Yoosee app and then the same from the Onvif Device Manager, trying to execute the same action. Could you please share them?

brunobittarello, 2025-03-24:
Here it is, I hope I did right because I am not familiar with wireshark
[yoosee.zip](https://github.com/user-attachments/files/19430775/yoosee.zip)

brunobittarello, 2025-03-24:
> So in the yoosee.pcap there is no Onvif. Perhaps the app is using a different protocol to talk to the camera?
I used the command netstat -bn to check what Yoosee is using. It is only targeting 127.0.0.1 (50521, 50522 and 50524). I dont know how I can detect if it using something else. Do you know how I can get this?

> In the Onvif Device Manager, I can see you are using "Continuous Move" (bottom right radio button). Try changing it to "Relative Move". The way Relative Move works is that by pressing down and holding the button the camera starts moving, releasing the button stops the movement. Cheap Chinese cameras usually don't have the more expensive stepper motors which is why they don't support Absolute/Continuous movement and sometimes they even lie about this capability in the response.

I didn't see that option, but they both didn't move this time.

Besides Hikvision cameras, do you recommend any other camera brand?

jimm98y, 2025-03-24:
Well, I wouldn't say I "recommend HikVision", it's just that I happen to have access to a few of them and they seem to support the standard pretty well. In general, any camera that supports Onvif should do fine. 

This project was started primarily as the Onvif server to allow you to build Onvif compatible camera, or add Onvif support for an existing camera without Onvif. The client part was created in order to test the server, not having to rely upon the Onvif Device Manager.

closed 2025-03-24 by jimm98y

brunobittarello, 2025-03-24:
I see, I am going to buy a better camera and try to access it, it will be better than "fix" or doing workaround with this two cameras. Thanks for your time and attention!