jimm98y/SharpOnvif #20: Sample of recordings

carlos-chourio, 2025-03-25:
Hello Jimm, one question do you by any chance have a sample of creating a video recording on a specific camera using the library? 

I am trying to create one recording and then retrieve it later using the Search.

I have tried something like this but does not work, return a 400 bad request:

public async Task<string> CreateRecording(Profile cameraProfile)
{
    string recordingURL = await GetServiceUriAsync(OnvifServices.RECORDING).ConfigureAwait(false);
    var recordingClient = GetOrCreateClient<RecordingPortClient, SharpOnvifClient.Recording.RecordingPort>(recordingURL, (u) => new RecordingPortClient(OnvifBindingFactory.CreateBinding(), new EndpointAddress(u)));
    var recordingToken = await recordingClient.CreateRecordingAsync(

        new RecordingConfiguration()
        {
            
            Content = "Video",
            MaximumRetentionTime = "PT10S",
            Source = new Recording.RecordingSourceInformation()
            {
                SourceId = cameraProfile.token,
                Name = cameraProfile.VideoSourceConfiguration.Name,
            }
        }).ConfigureAwait(false);
    return recordingToken;
}

I am specifically passing it a camera profile of a camera I have connected to the recorder but with no luck.

 Btw I saw that the find recording according to onvif docs can contain StartPoint and EndPoint although your implementation does not have them. I added them to try it out and check if that works but since I haven't been able to create recordings I still haven't tested that. 

If I manage to make it work maybe I could make a PR later on.

Thank you in advance for your response, and thank you for your great work in this library

jimm98y, 2025-03-25:
FindRecordings as defined in https://www.onvif.org/ver10/search.wsdl which was used to generate this client does not contain StartPoint/EndPoint, which explains why it's missing. 

As for an actual sample, I'll have to first try it on my cameras - at a glance I don't see anything wrong with your code.

carlos-chourio, 2025-03-27:
Oh, ok. Thank you very much! I will try with my recorder/camera although I think it's kinda broken but I will take a look. Hopefully I can make it work

jimm98y, 2025-03-29:
So I've set up recording on my cameras, but it turned out this particular action returns "This optional method is not implemented." To me it looks like a correct result that is coming from the camera, indicating that this particular optional method was not implemented by the manufacturer. For reference, here is the code I used for testing:
```cs
var cameraProfiles = await GetProfilesAsync();
string recordingURL = await GetServiceUriAsync(OnvifServices.RECORDING).ConfigureAwait(false);
var recordingClient = GetOrCreateClient<RecordingPortClient, SharpOnvifClient.Recording.RecordingPort>(recordingURL, (u) => new RecordingPortClient(OnvifBindingFactory.CreateBinding(), new EndpointAddress(u)));
var recordingToken = await recordingClient.CreateRecordingAsync(
    new RecordingConfiguration()
    {
        Content = "Video",
        MaximumRetentionTime = "PT60S",
        Source = new RecordingSourceInformation()
        {
            SourceId = cameraProfiles.Profiles.First().token,
            Name = cameraProfiles.Profiles.First().VideoSourceConfiguration.Name,
        }
    }).ConfigureAwait(false);
```

closed 2025-06-09 by jimm98y