jimm98y/SharpOnvif #21: Special handling for devices that don't send 100 Continue

JTimothyKing, 2025-03-25:
I've encountered a strange behavior with a couple of cameras. The ONVIF client starts the HTTP connection by including an `Expect: 100-continue` in the HTTP 1.1 request headers. This header indicates that the client has potentially a lot of data to send in the request body and wants to make sure that the server (the device) is ready to receive it. At this point in the sequence, the device is expected to reply with `100 Continue`, after which the client sends the HTTP request body. Most devices seem to behave as expected.

However, I've run across a couple of cameras that never send `100 Continue` in reply. The client seems to wait for 1 second and then send the request body. This is an issue, especially with continuous PTZ commands, where my software receives direction from the user (e.g., from a UI) and then needs to send the command to the camera in real-time. Inserting a 1-second delay for each HTTP round trip adds a noticeable lag in the UI. (Additionally, HTTP Digest authentication requires 2 round trips.)

I'm looking to disable `Expect: 100-continue` or to reduce the delay, and it looks like that requires delving deeply into the WCF binding code.

JTimothyKing, 2025-03-25:
There is a global property `ServicePoint.Expect100Continue`, but that doesn't appear to have any effect. This property is documented as obsolete, and I'm guessing that it is no longer respected in recent WCF code.

I'm continuing to delve into this, and if I discover or implement a solution, I'll post it here.

jimm98y, 2025-03-25:
I haven't experienced the issue myself, but this looks promising: https://justsimplycode.com/2019/11/02/disable-header-100-continue-in-net-core-wcf-client/
I suppose I could make it optional...

PR #22 (2025-03-25): F disable 100-continue
Closes #21
commit fb641224 (2025-03-25): F Added disable 100 continue option to fix #21
closed 2025-03-25 by jimm98y