jimm98y/SharpOnvif #25: Can't authenticate Network Optix's Software

StefanoNDP, 2025-05-16:
Hello, I'm new in this Onvif world and I'm trying to make it work in nxvms' software, unfortunatelly it won't authenticate and I don't know why.

Steps to reproduce:
1: Clone this repo
2: Open OnvifService in Visual Studio
3: Run OnvifService
4: Try to connect from nxvms software
5: Get errors below.

<details>
  <summary>Click to expand</summary>
info: CoreWCF.Channels.ServiceModelHttpMiddleware[0]
      Mapping CoreWCF branch app for path /onvif/device_service
info: CoreWCF.Channels.ServiceModelHttpMiddleware[0]
      Mapping CoreWCF branch app for path /onvif/media_service
info: CoreWCF.Channels.ServiceModelHttpMiddleware[0]
      Mapping CoreWCF branch app for path /onvif/events_service
info: CoreWCF.Channels.ServiceModelHttpMiddleware[0]
      Mapping CoreWCF branch app for path /onvif/Events/SubManager
info: CoreWCF.Channels.ServiceModelHttpMiddleware[0]
      Mapping CoreWCF branch app for path /onvif/ptz_service
info: CoreWCF.Channels.MetadataMiddleware[0]
      Configuring metadata to /onvif/device_service
info: CoreWCF.Channels.MetadataMiddleware[0]
      Configuring metadata to /onvif/media_service
info: CoreWCF.Channels.MetadataMiddleware[0]
      Configuring metadata to /onvif/events_service
info: CoreWCF.Channels.MetadataMiddleware[0]
      Configuring metadata to /onvif/Events/SubManager
info: CoreWCF.Channels.MetadataMiddleware[0]
      Configuring metadata to /onvif/ptz_service
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: https://REDACTED:5001
info: Microsoft.Hosting.Lifetime[14]
      Now listening on: http://REDACTED:5000
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Development
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\REDACTED\SharpOnvif\src\OnvifService
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[12]
      AuthenticationScheme: Digest was challenged.
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found
info: SharpOnvifServer.DigestAuthenticationHandler[7]
      Digest was not authenticated. Failure message: No authentication found  
</details>

Any help would be appreciated.

Note: Works with ODM without problems and works with HappyTime Onvif Client with a few "Not Implemented" Methods (just pressing to "Continue" on all issues and it works).

jimm98y, 2025-05-16:
It complains about the incoming requests missing authentication. Did you configure your credentials in nxvms? Could you share Wireshark logs? 

"works with HappyTime Onvif Client with a few "Not Implemented" Methods"
Which methods are throwing NotImplementedException? I could perhaps implement them to make the sample work out of the box.

StefanoNDP, 2025-05-16:
When I get home I'll send the wireshark logs, regarding the methods those are:

From MediaBase:

	GetAudioEncoderConfigurationsResponse
	GetVideoEncoderConfigurationsResponse
	GetVideoSourceConfigurationsResponse
	
from PTZBase:

	GetConfigurationsResponse

jimm98y, 2025-05-16:
Thank you! I've added the methods into the sample app, please get the latest version and try it out. Onvif Device Manager does not call them, so I can't test if it's working correctly.

StefanoNDP, 2025-05-19:
Good morning! Thanks for the reply.

GH doesn't allow wireshark's pcapng files so I exported them as .txt and .csv, if you want the pcapng I can DM it to you or upload somewhere and link it to you.

[nx_sharponvif_packets.txt](https://github.com/user-attachments/files/20279574/nx_sharponvif_packets.txt)
[nx_sharponvif_packets_all.csv](https://github.com/user-attachments/files/20279575/nx_sharponvif_packets_all.csv)

jimm98y, 2025-05-19:
Unfortunately, the content of the messages I was looking for is missing in the text dumps. Could you send the pcapng dumps to [asdasffqfwefefeffe@outlook.com](mailto:asdasffqfwefefeffe@outlook.com) please?

StefanoNDP, 2025-05-20:
Sent

jimm98y, 2025-05-20:
Thank you! Unfortunately, all seems to be OK in the logs. There are some requests outside of Onvif that are failing (ISAPI and CGI), but Onvif seems to work correctly using the Ws-UsernameToken auth.

See for instance towards the end of the capture packet 464 (POST to /onvif/device_service endpoint), returns 401 Unauthorized in packet 466, then the next request 476 has Ws-UsernameToken and succeeds, getting OK in 478.

Is it possible that nxvms fails to identify the camera (due to CGI or ISAPI calls failing) and shows error? What is the error you get in nxvms?

StefanoNDP, 2025-05-21:
It doesn't show any error, it just shows its loading bar and then after a few seconds it stops.

It also doesn't discover automatically the device, I have to manually click on "Add device" and manually write the info there.

jimm98y, 2025-05-22:
As for the discovery, it is strange that the service does not respond. I've tried to parse the discovery message from Wireshark and it did seem to work correctly, I can also see the message being sent to the correct multicast group. Perhaps it could be not discoverable due to some networking configuration (e.g. your Firewall blocking the ports), can you please try disabling the Windows Firewall and trying again? I also noticed all the errors were traced only using System.Diagnostics.Debug, so I've enhanced logging/tracing to use the ILogger - please get the latest changes from this repo. When running outside of Visual Studio, please set appsettings.json/Logging/LogLevel/Default to "Debug" in order to see all the received requests and responses. Errors should now pop up at startup in the console. 

As for the service not being visible in nxvms, perhaps it can be due to some response that it did not like - the last one in the Wireshark traces was about the NetworkInterface where in your case it's detecting (most likely) wrong network IP "169.254.21.45" and the HW address is empty - you might want to try tweaking the DeviceImpl.cs/GetNetworkInterfaces method to return your primary IP/MAC address and see if it makes a difference. Running the OnvifServer as Administrator (either run Visual Studio as Administrator, or start OnvifService.exe as Administrator by right clicking the EXE and selecting "Run as Administrator") might help you as well.

StefanoNDP, 2025-05-23:
I've got good news and bad news!

The good news: Thanks to you I've managed to make it recognize the device, by writting both the Ethernet Adapter Name and MAC Address, it managed to find the device a connect to it.

The bad news:
1. It doesn't discover the device by itself
2. It complains about a bunch unimplemented methods
3. It doesn't use the address I provide for rtsp.

1*: I have to manually write the address:port and credentials

2*: The methods are:
From MediaBase.cs
```
  GetVideoSourcesConfigurationOptionsResponse
  GetCompatibleVideoEncoderConfigurationsResponse
  GetCompatibleVideoDecoderConfigurationsResponse
  GetAudioOutputConfigurationsResponse
  AudioEncoderConfigurationOptions
  SetVideoEncoderConfiguration
  AddPTZConfiguration
  AddAudioOutputConfiguration
  AddAudioDecoderConfiguration
```

From DeviceBase.cs
```
  GetRelayOutputsResponse
  SetRelayOutputState
```

From PTZBase.cs
```
  Capabilities
```

3*: I'm using the rtsp from another IP address aside from my own local ip address, 
Example: 
My local ip is 192.168.2.111 and the rtsp is 192.168.1.222

If I put rtsp://192.168.1.222:554/myvideo.ext in MediaImpl.cs, the program tries to connect to rtsp://192.168.2.111:443/myvideo.ext

jimm98y, 2025-05-23:
Missing functions were added, but again I cannot test them - please let me know if it works now.

As for the discovery, did you get the latest version? Were there any errors related to the DiscoveryService in the output?

jimm98y, 2025-05-24:
"If I put rtsp://192.168.1.222:554/myvideo.ext in MediaImpl.cs, the program tries to connect to rtsp://192.168.2.111:443/myvideo.ext"
Well, if you do that, then nxvms gets that URL and it's up to them what they do with it. The behavior you described seems like some security feature where they discard the host and always assume the camera provides both the Onvif and video, so perhaps a workaround would be running your own RTSP proxy (this might help, but some work will be needed to create the proxy https://github.com/jimm98y/SharpRealTimeStreaming) locally on the same machine as SharpOnvif and let it forward the stream from the other machine. Or use existing solutions such as live555proxy to do the forwarding for you.

StefanoNDP, 2025-05-26:
I've cloned your repo again, and:

Its complaining about 2 unimplemented functions:

  From DeviceBase.cs
    `SetRelayOutputSettings`

  From PTZBase.cs
    `GetCompatibleConfigurationsResponse`

Regarding RTSP on nxvms, I can confirm that it only checks for what's after the address.

I've tested address:port/my.video, address:port/mystream, address:port/test.sdp, address:port/video.mp4, etc (where address is rtsp://192.168.1.222:554) and it was always looking for deviceaddress:port/my.video, deviceaddress:port/mystream, deviceaddress:port/test.sdp,  deviceaddress:port/video.mp4, etc (where deviceaddress is rtsp://192.168.2.111:554)

And about discovery, I've used wireshark and I think it's looking for the "anonymous" user, I've sent an e-mail to [asdasffqfwefefeffe@outlook.com](mailto:asdasffqfwefefeffe@outlook.com) with the wireshark dump about it

jimm98y, 2025-06-02:
"EDIT: It's only discoverable on ODM, all the others (HappyTime's Software, Network Optix's NXVMS, DigiVOD, etc) I have to manually add SharpOnvif Server's info to connect."

Ok, that is interesting. Would you have some Wireshark capture of the discovery from the other (non-ODM) apps where it shows the probe and a valid response from devices that you can discover? It could be an issue in my code, I was only able to develop/test using ODM.

StefanoNDP, 2025-06-03:
E-mail sent with the dumps.

jimm98y, 2025-06-03:
Thank you for the pcap!

DigiVOD.pcap does not show Onvif discovery, only M-SEARCH (UPnP). That would explain why it's not working.

nxvms.pcap unfortunately does not show SharpOnvifServer->nxvms direction, I can only see traffic from nxvms->SharpOnvifServer. It looks like it's sending 3 Onvif probes for:
1) dp0:NetworkVideoTransmitter
2) dp0:NetworkVideoDisplay
3) dp0:Device
Of those 3 probes, dp0:NetworkVideoDisplay is not supported by SharpOnvifServer, dp0:Device is sent with incorrect namespace (http://www.onvif.org/ver10/network/wsdl while we are looking for http://www.onvif.org/ver10/device/wsdl). However, the third probe for dp0:NetworkVideoTransmitter looks correct to me and should have received a response from SharpOnvifServer. Can you please capture it again, this time also including outgoing UDP traffic from SharpOnvifServer into the network? Also did could you please attach all the output from SharoOnvifServer console window please?

happytime.pcap also looks OK, please include the direction from SharpOnvifServer into the network and the console output as well.

jimm98y, 2025-06-03:
I've just confirmed it. To rule out your network (firewall, etc.) here is a simple test you can try:

1) clone this repo
2) open and run SharpOnvifServer on localhost, don't make any changes
3) modify the SharpOnvifClient/OnvifDiscoveryClient.cs, line 88 - delete the entire message and copy/paste the XML from your Wireshark capture nxvms.pcap => dp0:NetworkVideoTransmitter or any from the happytime.pcap. 
4) compile it and run it
5) it will find the server and connect.

If you try the invalid ones, such as the dp0:Device from nxvms.pcap, you will get no response and the Client will exit, prompting you to check your permissions.

StefanoNDP, 2025-06-04:
Done all of this, still nothing. VS Running as Admin, Firewall off, replaced the entire XML message from OnvifDiscoveryClient, changed AddOnvifDiscovery function to match yours above, still none but the ODM could discover it automatically.

And I couldn't connect to NXVMS without manually changing the network interface's name and mac address.

The filter I used were including UDPs as well (there just wasn't any). And about I don't think I've changed anything regarding in/out connections.

I've tested each XML with both 

```c#
builder.Services.AddOnvifDiscovery();
```

and 

```c#
builder.Services.AddOnvifDiscovery(new SharpOnvifServer.Discovery.OnvifDiscoveryOptions() { 
 Types = new System.Collections.Generic.List<System.Tuple<string, string>>()
 {
     new System.Tuple<string, string>("http://www.onvif.org/ver10/network/wsdl", "NetworkVideoTransmitter"),
     new System.Tuple<string, string>("http://www.onvif.org/ver10/network/wsdl", "Device")
 }
});
```

These were the XML I used
<details>
  <summary>XML1</summary>
            "<s:Envelope\r\n" +

            "  xmlns:s=\"http://www.w3.org/2003/05/soap-envelope\"\r\n" +

            "  xmlns:a=\"http://schemas.xmlsoap.org/ws/2004/08/addressing\">\r\n" +

            "  <s:Header>\r\n" +

            "     <a:Action s:mustUnderstand=\"1\">\r\n" +

            "     http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe\r\n" +

            "     </a:Action>\r\n" +

            "     <a:MessageID>\r\n" +

            "       urn:uuid:" + uuid + "\r\n" +

            "     </a:MessageID>\r\n" + // uuid has to be unique for each request

            "     <a:ReplyTo>\r\n" +

            "       <a:Address>\r\n" +

            "         http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous\r\n" +

            "       </a:Address>\r\n" +

            "     </a:ReplyTo>\r\n" +

            "     <a:To\r\n" +

            "       s:mustUnderstand=\"1\">\r\n" +

            "       urn:schemas-xmlsoap-org:ws:2005:04:discovery\r\n" +

            "     </a:To>\r\n" +

            "  </s:Header>\r\n" +

            "  <s:Body>\r\n" +

            "    <Probe\r\n" +
[... 130 lines cut ...]

            "        xmlns=\"http://schemas.xmlsoap.org/ws/2005/04/discovery\">\r\n" +

            "        <Types>\r\n" +

            "            dn:NetworkVideoTransmitter\r\n" +

            "            </Types>\r\n" +

            "        <Scopes/>\r\n" +

            "        </Probe>\r\n" +

            "    </Body>\r\n" +

            "</Envelope>\r\n";
</details>

This is the filter I used:

`ip.addr == 192.168.101.107 && ip.addr != 35.186.224.24 && ip.addr != 224.0.0.251 && ip.addr != 91.189.91.157 && !wg && ( http || xml || ssdp || udp )`

These other IPs are from microsoft, google, aws (but they change constantly...) and I had to filter out WireGuard (!wg) since it was overwhelming the log

Please, tell me step-by-step what I have to do on wireshark to get a good dump (lol).

jimm98y, 2025-06-04:
> Also, am I supposed to run both SharpOnvifServer and ShaprOnvifClient? Be cause when I do, the OnvifDiscoveryClient throws an error on line 110 "using (UdpClient client = new UdpClient(endPoint))", saying not a valid IP address.

Yes, at the same time, both should work on the same interface on localhost. That's how I usually debug it - run the server in one Visual Studio and the client in other Visual Studio, by default they should find each other and connect. Since you are saying you still have to patch the IP/MAC ("And I couldn't connect to NXVMS without manually changing the network interface's name and mac address"), perhaps seeing what's the original value might offer a clue on what's going on. 

I was thinking "ip.addr == 192.168.101.107 && udp" should be enough, but looking at your filter it looks ok to me. Then my current hypothesis is, that SharpOnvifServer takes a wrong network adapter at startup and it uses it for the UDP broadcast listener as well as the configuration response => the multicast never reaches the server because it's listening on a different network adapter. How many network adapters do you have?

StefanoNDP, 2025-06-05:
I have 7 network adapters:

    1 for wired/wireless connection (Active and being used)
    1 from wireshark (Active and being used)
    1 from wsl (Active and being used)
    1 from wireguard (Active and being used)
    1 from bluetooth network (Active but unused)
    2 from OpenVPN (TAP and Wintun) (Both active but unused)

jimm98y, 2025-06-05:
OK, having so many network interfaces explains it, I have just a single Ethernet adapter from Parallels. Looking into possible causes, it seems like UdpClient can only join a multicast group on a single network interface, not on all of them at the same time. And even though I am listening on all network interfaces (using the IPAddress.Any), apparently the multicast group is joined only on one of them based upon your Windows settings. This will require a bigger change in how the server responds to the multicast, I'll have to create a client for each network interface I suppose.

StefanoNDP, 2025-06-09:
I've tried the latest repo version, it's still not being discovered automatically but I can connect without having to change anything in the code.

StefanoNDP, 2025-06-11:
E-mail sent

jimm98y, 2025-06-11:
Thank you! You're right, there are no responses.... I wonder if it's the same issue as #26 - if you were using your wireless adapter for the discovery, it might explain it.

You can try getting the latest version, it should now also work on wireless adapters (and a few other NIC types as well). If not, I've just added a configuration in appsettings.json where you can specify the network interface you want to use for discovery:
```json
"OnvifDiscovery": {
 "NetworkInterfaces": [ "192.168.1.77" ],
}
```

StefanoNDP, 2025-06-13:
E-mail sent.

One thing I forgot to mention in the e-mail is that, if more than one NIC is enabled, I get Null Reference Exception from the
`return nic.GetIPProperties().GatewayAddresses.FirstOrDefault(x => x.Address.AddressFamily == System.Net.Sockets.AddressFamily.InterNetwork).Address?.ToString();`
in `GetPrimaryIPv4Gateway` method in DeviceImpl.

jimm98y, 2025-06-13:
Hopefully the exception is fixed now - it's likely that your NIC does not have a gateway assigned. I've added more checks and Gateway should now also be configurable from appsettings.json if needed.

StefanoNDP, 2025-06-16:
Yes, the exception is fixed. Thanks!

But it's still "finding" multiple SharpOnvif's servers.

StefanoNDP, 2025-06-20:
I don't think I've limited to only one direction.

How do I make it capture both directions?

StefanoNDP, 2025-07-16:
ODM: Works out of the box

NXVMS: Have to manually change localhost to an actual IP address in the launchSettings.json... To prevent the bug of finding the same device multiple times (show as different mac addressess) I have to manually change the "OnvifDiscovery:MAC" value from null 
to the actual MAC Adress from "appsettings.json". Aside from that, everything's working.

HappyTime and DigiVOD: None auto discovers, both connects when manually write SharpOnvif's info.

Also I've made some QOL (IMHO) changes so we can take care of things from just one place.

jimm98y, 2025-07-16:
Thank you! I did find the issue: it was not getting the default values at all, so IP/MAC were all empty strings unless configured in appsettings.json. Fortunately, the issue was just in the sample app, it does not affect the library. Please try it out in the latest version.

closed 2025-07-16 by jimm98y