jimm98y/SharpOnvif #29: Issue with GetProfiles request

ApecJoe, 2025-06-16:
Hello, I am currently trying to run the sample Onvif Server, however it isn't working. Im looking at the packets being sent out with wireshark and it seems as if everytime a request goes out for GetProfiles, the response is always a `500 Internal Server Error` and it has the following error code:

```The message with Action 'http://www.onvif.org/ver10/media/wsdl/GetProfiles' cannot be processed at the receiver, due to a ContractFilter mismatch at the EndpointDispatcher. This may be because of either a contract mismatch (mismatched Actions between sender and receiver) or a binding/security mismatch between the sender and the receiver.  Check that sender and receiver have the same contract and the same binding (including security requirements, e.g. Message, Transport, None).```

Any help would be greatly appreciated.

ApecJoe, 2025-06-17:
We are using Onvif Device Manager, and also a custom client that uses sharpOnvif, and it fails at GetProfiles on both of them.

I tried those steps, and it manages to connect to each other and work fine. I've then tried connecting to the server on our custom client, and it still fails at GetProfiles.

jimm98y, 2025-06-17:
I am using ODM 2.2.250 against the SharpOnvifServer and it looks like it works just fine when I select "Profiles" - it lists the main profile with all the settings. Did you make any changes to the code?

commit 7960669a (2025-06-17): D Documentation improvements - closes #29
closed 2025-06-17 by jimm98y